sinara-hw/Metlino #3: Metlino Ethernet

gkasprow, 2018-11-07:
@michallgaska

- Replace MAX24287 by M88E1111. This is same IC as KC705 which is already supported by ARTIQ. 
- FPGA configures M88E1111 -- no interface with MMC. 
- Layout so that uTCA Port0 and SFP can be driven by M88E1111 or by GTH transceiver on FPGA. Use coupling capacitors to choose which.
- Default stuffing and development path is M88E1111. GTH path is for risk mitigation.

closed 2019-04-24 by gkasprow

sbourdeauducq, 2019-05-18:
>     * Default stuffing and development path is M88E1111. GTH path is for risk mitigation.

Can we get GTH by default? GTH Ethernet is working on Sayma now, and doesn't require fiddling with PHY registers or messy GTH I/O clocking.

gkasprow, 2019-05-19:
sure.done.