sinara-hw/Metlino #4: Metlino P2P links

gkasprow, 2018-12-13:
For Metlino there are defined  IOSERDES links between main FPGA and Sayma FPGA GTH using second lane of Fat Pipe 1  (Fabrc E). Fabric D is used for GTY-GTY communication and DRTIO. On Sayma  second lane won't be connected to GTY anymore because that GTY would be routed to RTM FPGA.
Once we added WR support, I miss a few LVDS lines to implement all 12 links on Metlino.
The question is if we are ever going to use these lanes? Maybe we should just remove them all. At the moment they are distributed across the banks. Some reside within VHDCI signals, some in SDRAM bank, some among FMC lanes. Clocking such distributed IOSERDES would be tricky.
I have one HR bank connected at the moment to 3.3V, but could switch it to 2.5V to support such additional channel. Some level converters would be needed to support WR I2Cs. On Sayma we could connect the second lane of FatPipe to IOSERDES and gain additional  IOSERDES channel.
The question is - @sbourdeauducq @jordens @hartytp  is it worth to implement them?

jbqubit, 2018-12-13:
@gkasprow It sounds like you're asking multiple things at the same time. AFAICT...
1) If Sayma doesn't support Fabric E then it's not needed on Metlino. 
2) If Fabric E support is removed from Metlino do you have enough LVDS lines for WR on Metlino?

gkasprow, 2018-12-13:
Existing Sayma has fabric D and E connected to GTH. Sayma v2 will have one GHT assigned to RTM FPGA and another to fabric D. But I can easily add regular LVDS line to further support it.
After I added WR, I miss one LVDS line to support fabric E on Metlino for 12 slots. But such support would be problematic because individual lanes are scattered between banks with different VCC. I have one HR bank that has just a few signals used for slow control, I can assign them somewhere else and use this one bank (with VCC=2.5V) to fully support fabric E.
What are the plans to use such fabric?

gkasprow, 2018-12-14:
I'm publishing the pdf. Let's do the review. Once we decide what to do with these links, I will update the design.

gkasprow, 2019-04-24:
we don't care about theses links.

closed 2019-04-24 by gkasprow