sinara-hw/Metlino #16: MCH Fabric D and E need at least 1 harness connector in the schematics

marmeladapk, 2019-04-26:
Right now there are many errors caused by this, signals are not recognised as part of the harness.

Could possibly affect other harnesses in the project also.

gkasprow, 2019-04-26:
not all signals are routed to FPGA.
I would simply remove the fabric E since we cannot ensure connectivity for all Sayma slots

marmeladapk, 2019-04-26:
Yes, but the point is, that signals are to in the harness and thus are not connected accross schematics:

[image]

gkasprow, 2019-04-26:
that's strange. Do you have a harness file with the definition? It's required.
There are 2 ways of defining harness. Either you draw harness connector and the file is generated automatically or you define the harness in the file and use nets to keep connectivity. In the second case, the harness entry in the file is not generated automatically.

marmeladapk, 2019-04-26:
I compiled the project straight from the repository. Perhaps some files are missing and weren't created automatically?

marmeladapk, 2019-04-26:
WR_I2C_CTRL on MCH_FPGA_BANKS_64_65.schdoc for example doesn't register all signals MAIN_DCXO_OE, SCL, SDA are not shown as part of the harness.

marmeladapk, 2019-04-26:
On AD17 and higher it is registered correctly.

closed 2019-04-26 by marmeladapk