sinara-hw/Metlino #18: Daughterboard connector

marmeladapk, 2019-05-06:
@gkasprow 
- Was communication via SPI with main MCH removed on purpose?
- WRCLK has an additional connector for sending clock signal (J24) -  should Metliuno also have this connector and if not we should check if it doesn't interfere mechanically.

gkasprow, 2019-05-11:
SPI communication is only for ATMEGA update via MCH. Since we are using our own CPU, we don't need it.
@marmeladapk what do you mean by WRCLK and which J24?

marmeladapk, 2019-05-11:
@gkasprow Formerly WR-MCH and J24 on WR_MCH.

gkasprow, 2019-05-11:
In WR-MCH this connector is for PCIe clock only. We won't use PCIe here
And, this connector conflicts with SDRAM.
For this reason, it's easier to add ufl jumper that distributes needed clocks.
There is a use case for WR-MCH and Metlino for the NICA-MCORD project and we will need WR timing distribution but it is needed in AFCK boards only to time-stamp samples. Metlino would serve here as a data concentrator only. 
When needed, we can use ufl jumpers.

closed 2019-05-11 by gkasprow