sinara-hw/Metlino #31: Metlino MMC

gkasprow, 2019-05-07:
We asked NAT to develop firmware that allows correct powering of the WR-MCH module from NAT MCH. We could do the same for Metlino. @saleksin takes care of it in Creotech.
Such an approach would let us remotely update the MMC from MCH side and report the presence of Metlino. And of course, enable power supply and Ethernet for it :)
But we would probably have to use the ATMEGA CPU because such chip is used as MMC for NAT MCH mezzanines.

jbqubit, 2019-05-09:
Configuring Metlino to be always-on and updating Metlino MMC via header is fine. Seems like adding a new CPU, tool chain and 3rd party code is a big change for small convenience gains.

gkasprow, 2019-05-10:
This is true providing that we get from NAT info what we need to do to get the power and Ethernet.
NAT offered to give us binaries that we simply burn in the CPU to get power and connectivity.
This, of course, makes life easier for us but would be the only part that is a black box. On the other hand, the NAT MCH is a black box either so personally, I'd choose such an approach. When we make our own MCH board in the future, the power can be enabled by default so such CPU won't be needed.

sbourdeauducq, 2019-05-19:
NAT firmware and software doesn't exactly have a track record of quality, so likely this would be a stopgap at best.
IMO the best way to deal with microTCA in the long term is to take the good things from NAT (basically some hardware: crate, power supply module, fans), and replace the bloated MCH and its crappy firmware with one little board with a STM32 MCU and an integrated Ethernet switch-on-a-chip.
At the same time we can probably ditch most of IPMI and only implement the bare minimum so that the power supply and the fans behave themselves.

gkasprow, 2019-05-19:
@sbourdeauducq we will do so sooner or later.
I got info from NAT that we don't really need to talk to MCH. One need to declare more power that MCH requests from PSU. Ethernet is enabled in an independent way.
Power is on by default.
So we will keep the existing chip and use it for power sequencing nad monitoring.

closed 2019-05-19 by gkasprow