sinara-hw/Metlino #38: stop calling everything "WR"

sbourdeauducq, 2019-05-19:
The Si5324 has nothing to do with White Rabbit.
[image]

What about calling this (Si5324, DCXOs, DDMTD flip-flops) the "central PLL" or the "central clocking subsystem"?

sbourdeauducq, 2019-05-19:
Also here:
[image]
Why is that a "WR" reference clock?
By the way, for clocks to be usable for DRTIO they need to go to transceiver clock inputs, GC pins cannot clock the transceivers well and can only be used to clock local RTIO peripherals. I am not sure what the use case is for this clock, and I don't know if this is a problem or not.

hartytp, 2019-05-19:
Yes we should stop calling things wr as there is much more to the wr stack than a ddmtd pll. I don’t want to bikeshed terminology and will be happy with anything sensible and consistently applied.

hartytp, 2019-05-19:
Was that ref the ddmtd pll reference when Metlino is used as the drtio master?

sbourdeauducq, 2019-05-19:
Why do we need DDMTD when it's used at the master?

hartytp, 2019-05-19:
True if you won’t use the DCXO at all then it’s not needed. On Kasli we’d want it to provide a nice clean clock to the EEMs but Metlino doesn’t have that requirement

gkasprow, 2019-05-19:
OK, now I call this harness Clocking_CTRL and removed "WR" from signal names.

closed 2019-05-19 by gkasprow