sinara-hw/Metlino #39: position of GC pins on VHDCI connector

sbourdeauducq, 2019-05-19:
The VHDCI connector is supposed to be used with the VHDCI carrier, but for this purpose the location of GC pins does not seem to match.
For example, pins 1/35 of the VHDCI0 connector do not go to a GC pin of the Ultrascale FPGA, but connect to bit 0 of EEM0, which is used as a clock signal.

gkasprow, 2019-05-19:
fixed. Now pins 1/35, 2/36, 20/54, 21/55 go to GC
[image]

sbourdeauducq, 2019-05-20:
On 1.0rc5, FPGA pin of EEM bit0 for...
EEM0: D25/E25
EEM1: C22/C21
EEM2: A25/B25
EEM3: B26/C26
EEM4: W24/W23
....
So far only EEM0 and EEM4 are correct (go to GC pins). Or am I missing out something?

closed 2019-05-20 by sbourdeauducq