sinara-hw/Metlino #58: PI analysis after optimization

gkasprow, 2019-07-01:
12V rail. Drop is mainly on MOSFETs.
[image]
in one place the current density is high on the bottom layer. But we cannot do much. Via currents are fine.
[image]

gkasprow, 2019-07-01:
1V rail
[image]
[image]

gkasprow, 2019-07-01:
1V5 rail. The DC drop includes also 1V2 LDO so it is fine.
[image]
[image]

gkasprow, 2019-07-01:
1V8
[image]
[image]

gkasprow, 2019-07-01:
3V3 rail. Current density is slightly too high, but on the top layer polygon, so it's fine.
[image]
[image]

gkasprow, 2019-07-01:
VCCINT. Note that the drop includes the coil DCR. In our case the voltage is stabilized under the FPGA.
Current density is high but we cannot really do much with it. It is dominated by buck chip to inductor connection.
[image]
[image]

gkasprow, 2019-07-01:
5V0 rail. It is used to supply 3V3 to clocking circuit via LDO. Simulation assumes that 3.3V LDO was shorted and includes 5V buck coil.
[image]
[image]

gkasprow, 2019-07-01:
P3V3MP
[image]
[image]

closed 2019-07-01 by gkasprow

jbqubit, 2019-07-08:
Thanks @gkasprow!