sinara-hw/Metlino #59: Si analysis

gkasprow, 2019-07-01:
Eye diagram @12Gbit 65/66 before optimization of the edge connector vias and before removing NFPs :
[image]

gkasprow, 2019-07-01:
With NFPs removed and anti-pads increased
red - 2mm anti-pads
light blue - 1.6mm anti-pads
[image]

gkasprow, 2019-07-01:
Eye diagram for Tx lanes after heavy optimization. There are additional vias and couplig caps
[image]

gkasprow, 2019-07-02:
SFP loopback test with  SFP cage SI model. Settings at 12Gbit/s, 64/66
[image]
[image]

gkasprow, 2019-07-04:
SiDDR test after optimization finally passes for 1.066GT/s speed.
The detailed report is [here](https://www.dropbox.com/s/k3h3jbyh6trdmn2/DDR_Results_Jul-4-2019_8h-15m.zip?dl=0)

gkasprow, 2019-07-04:
[here](url)

closed 2019-07-04 by gkasprow

jordens, 2019-07-04:
The trace in https://github.com/sinara-hw/Metlino/issues/59#issuecomment-507141566 looks bad/weird. The difference hovers around 0 for half the UI. Was that an intermediate result or the final one?

gkasprow, 2019-07-04:
It was an intermediate TDR result. The eye diagram looks much better - similar to the last one. The problem is with 0.6mm connector vias because their impedance cannot really be matched and will always be 30Ohm in the best case.