sinara-hw/Metlino #60: manufacturing and testing status

jbqubit, 2019-12-09:
@marmeladapk When we spoke on 11/27 you said: 
- Metlino (qty 4) are stuffed but not yet tested. What is status of testing?
- Metlino-MCH Riser card is on hand. Do you have four of them?
- Front panel for Metlino + MCH is not yet on hand. Has it arrived yet?

jbqubit, 2019-12-16:
@marmeladapk Ping

marmeladapk, 2019-12-18:
- First Metlino card was powered up and works so far. Now we can finish developing our tests. I got info that some of them had been stuffed with LVDS version of Si549 due to CMOS version not being available in that quantity at the time of order).
- We have our own riser card, we got it with our MCH.
- Metlino panel prototype has been fitted, some small adjustments are needed. Afterwards it will be sent to manufacturer (similar timeline to Sayma panels).

kaolpr, 2019-12-19:
I did IBERT test for MGT on one of our cards and they look good.

- IBERT generated for all MGT at 10 Gbps
- Clocks used:
  - MGT228_0 (200 MHz, OSC2)
  - MGT226_0 (125 MHz, OSC3 -> IC16 -> MGT)
- Pins to configured:
  - Main_DCXO_OE (AM11) = 1
  - CLK_SEL (AM12) = 0
- MGT tested with two additional modules:
  - Internaly developed Metlino Tester (basically loopback for AMC signals)
  - [FMC Loopback  by IAM Electronic](http://www.iamelectronic.com/shop/produkt/fpga-mezzanine-card-fmc-loopback-module/)

**Results:**
 [image]
  - all links are up, besides of Link 19 (MGT 228/3) which was not looped back (Port 0).
  - ost of the links operates flawlessly at PRBS 31b and default settings
  - FMC links (MGT227 2, 3) are experiencing increased BER at PRBS > 9b.

marmeladapk, 2019-12-19:
@kaolpr Did you test FMC links with loopback or with FMC tester?

kaolpr, 2019-12-19:
>     * FMC Loopback  by IAM Electronic

marmeladapk, 2019-12-19:
It's a two layer board without any impedance control. I wouldn't use it for high speed signals.

kaolpr, 2019-12-19:
Yep, but it's configuration-free ;-) and if we use FMC tester there is ADN4604 in the signal path that has data rate limit at 4.25 Gbps.

marmeladapk, 2019-12-19:
What I meant is that increased BER doesn't mean that there's something wrong with Metlino.

kaolpr, 2019-12-19:
Totally agree, but indeed I could express it clearly.

kaolpr, 2019-12-19:
BTW, [similarly to Sayma AMC](https://github.com/sinara-hw/Sayma_AMC/issues/126#issuecomment-533541690), applying  DNP to R13 makes flash operational.

jbqubit, 2019-12-19:
Please post high-resolution photos of front & back side of a stuffed Metlino. Also post another photo showing all the stuffed Metlino arrayed on a table top. This is necessary so Anna can bill for production.

kaolpr, 2020-01-22:
I have just finished testing links in the crate and it looks that all links work. 

Metlino was mounted on NAT MCH, did get power supply (we're in contact with NAT to verify that it will work with every NAT MCH) and all backplane links work at 5 Gbps with Sayma AMC. Test was performed using Xilinx IBERT.

I'll do some more testing next week.

jbqubit, 2020-01-22:
> finished testing links

Did you try simultaneous links between rack of Saymas and Metlino at 5 Gbps? Isolated point-to-point isn't sensitive to eg cross-talk.  

@kaolpr Please post photos of the stuffed Metlino PCBs' front and back sides.

kaolpr, 2020-01-22:
Haven't tried multiple Saymas yet. I'm planning to do it on Monday.

Where do you want to have these photos posted?