sinara-hw/Metlino #61: Si549 populated with LVDS not CMOS

jbqubit, 2019-12-19:
> I got info that some of them had been stuffed with LVDS 
>version of Si549 due to CMOS version not being available in 
>that quantity at the time of order

It's unclear to me the implications of this change. @gkasprow ? Regardless, all Metlino should be stuffed identically to minimize board-to-board variation.

gkasprow, 2019-12-19:
Sure, providing that the oscillators are available. AFAIK they are software-compatible.

jbqubit, 2019-12-20:
@marmeladapk Have they been ordered?

jbqubit, 2020-01-22:
@gkasprow @marmeladapk Can this be closed?

marmeladapk, 2020-01-24:
Let's leave it open and close when they all have the same version of Si549.

sbourdeauducq, 2020-02-04:
The LVDS/CMOS capacitor selection is also wrong on some boards.

jbqubit, 2020-02-04:
What is s/n of board on which you see the problem?

marmeladapk, 2020-02-04:
@jbqubit See https://github.com/sinara-hw/Metlino/wiki/Prototype-reworks

All boards have capacitors mounted for CMOS version.

We plan to switch all boards to LVDS.

marmeladapk, 2020-02-21:
All boards now have LVDS version of Si549. Now I just need to find a moment to swap all capacitors and resistors to LVDS variant.

marmeladapk, 2020-03-04:
All boards have LVDS RC variant now.

closed 2020-03-04 by marmeladapk