sinara-hw/Metlino #65: Too high heatsink

kaolpr, 2020-01-21:
We've assembled Metlino with N.A.T. MCH and tried to put it in the crate ...

[image]

It looks like the heatsink is too high.

In case of the full-size create it is rather inconvenience that is limiting in some ways, but does not exclude use of Metlino. We can just leave second PSU slot unoccupied. 

Unfortunately, in case of R5 crate there is only one PSU slot and MCH slot is next to it so Metlino is mutually exclusive with PSU. Bad.

We can try to use a different heatsink that would have a lower profile, f.e. [ATS-60004-C1-R0](https://eu.mouser.com/datasheet/2/596/TS-60004-C1-R0-1480478.pdf) or [ATS-60002-C1-R0](https://eu.mouser.com/datasheet/2/596/TS-60002-C1-R0-1480509.pdf). 

Then we need to be extra cautious with temperature and verify if in an expected use-case we're able to maintain acceptable FPGA temperature.

gkasprow, 2020-01-21:
This heatsink needs machining. I can do that at the WUT.

kaolpr, 2020-01-22:
But what exactly do you mean by machining? 
Machining fins won't help as mounting mechanism will still collide.

gkasprow, 2020-01-22:
We can replace it with screws.

Pozdrawiam,
G.Kasprowicz

kaolpr, 2020-01-22:
We can try that, I'll bring one heatsink to WUT tomorrow.