sinara-hw/Metlino #67: SFP/USB plug collision

sbourdeauducq, 2020-02-03:
[image]

jbqubit, 2020-02-03:
SMA J1 (FRONT_SMA_CLK_IN) is OK
SMA J2 (SMA_CLK_OUT) collides with SFP J8A

@marmeladapk Consider replacing J2 SMA on prototype hardware with short SMA/SMP connector perpendicular to the PCB.

sbourdeauducq, 2020-02-03:
I think the SMA is fine (but double check) this issue is about the USB programming cable that definitely collides with SFP transceivers.

gkasprow, 2020-02-03:
we cannot do much here. We can move it under SFP so it blocks only one cage. Or move the USB to the backplane side.

jbqubit, 2020-02-03:
Looks to me like that SMA J2 also collides with SFP J8A. No?

jbqubit, 2020-02-03:
Would a very slim micro-USB cable fit? https://www.newegg.com/p/181-02AU-000X8

gkasprow, 2020-02-03:
Can you measure it with calipers?

sbourdeauducq, 2020-02-04:
We can also remove one of the SFPs and replace it with e.g. a SATA connector. What is the use case for having 3?

sbourdeauducq, 2020-02-04:
>  Or move the USB to the backplane side.

No. We do need to be able to reflash and debug Metlino in the crate in an easy manner.

jbqubit, 2020-02-04:
Use case for three SFPs on Metlino is same as on Kasli. 

Looks like Amazon has even thinner microUSB cables. https://www.amazon.com/Ribbon-Cable-Degree-Standard-Charging/dp/B0784M2X2C