sinara-hw/Metlino #68: Si5324 I2C low level is too high

marmeladapk, 2020-02-03:
Si5324 seems not to be able to properly pull SDA low. Tested with FPGA and with MMC talking to Si5324. Other than that Si5324 is reacting properly.

We modified pullup resistors on both sides to 10k. It didn't help.

Screenshot on CPU pullup resistors:

[image]

Screenshot on I2CMUX4 bus:

[image]

gkasprow, 2020-02-03:
It looks like one of the ICs was switched to fast mode (push-pull IOs) or MMC has wrong IO mode ( push-pull enabled )

gkasprow, 2020-02-03:
observe SDA before and after the I2C mux

kaolpr, 2020-02-04:
> Si5324 seems not to be able to properly pull SDA low. Tested with FPGA and with MMC talking to Si5324. Other than that Si5324 is reacting properly.
> 
> We modified pullup resistors on both sides to 10k. It didn't help.
> 
> Screenshot on CPU pullup resistors:
> 
> [image]
> 
> Screenshot on I2CMUX4 bus:
> 
> [image]

@gkasprow These are signals before mux (CPU pullup resistors) and after (I2CMUX4).

marmeladapk, 2020-02-04:
I shorted FPGA bus with Si5324 bus, ACK is still >1V. Both muxes are held in reset.

marmeladapk, 2020-02-04:
Just as a precaution I removed muxes and left only 10kOhm pullup (one per line). Still the same.

marmeladapk, 2020-02-04:
Found it!

FPGA I2C bus is also connected to the CPU (LPC1776).

CPU was configured differently than on Sayma: `IOCON_MODE_INACT` on Metlino, `IOCON_OPENDRAIN_EN` in Sayma. After changing this, Si5324 works in ARTIQ as @sbourdeauducq tested.

closed 2020-02-04 by marmeladapk