sinara-hw/Metlino #69: photos of Metlino

jbqubit, 2020-02-04:
Please post photos of Metlino front side, back side, front end-on, back end-on. 

https://github.com/sinara-hw/Metlino/issues/67#issuecomment-581549618

kaolpr, 2020-02-24:
### 1907180PW0006A0101

[image]
[image]
[image]
[image]

### 1907180PW0006A0102

[image]
[image]
[image]
[image]

### 1907180PW0006A0103

[image]
[image]
[image]
[image]

### 1907180PW0006A0104

[image]
[image]
[image]
[image]

### 1907180PW0006A0105

[image]
[image]
[image]
[image]

jbqubit, 2020-02-24:
At long last! Thanks.

closed 2020-02-24 by jbqubit

akaminska, 2020-04-07:
Unpopulated PCBs
[image]

jbqubit, 2020-04-07:
Thank you for the PCB photos. 

Do you also have Metlino photo with front panel installed? With riser card connecting to NAT MCH?

akaminska, 2020-04-07:
@kaolpr is preparing a photostory on how to integrate Metlino with MCH and attach panel right now. Should appear soon.

akaminska, 2020-04-16:
[image]
[image]
[image]