sinara-hw/Metlino #72: Possible typo in I2C address mapping

HarryMakes, 2020-08-25:
On page 27 of the latest version of [Metlino's schematic](https://github.com/sinara-hw/Metlino/releases/download/v1.0rc8/Metlino.PDF), there's a table showing all the different ports and addresses of the I2C slave devices. About the default addresses of the two TCA9548A switches, a paragraph reads:

> MMC I2C1, addr 1110 A2 A1 **0**
FPGA I2C and MMC I2C2, addr 1110 A2 A1 **1**
where A2=P2[28], A1=P2[29]
default addr is 0x70 and 0x71

However, the connection for IC43, which is the I2C switch connected to the FPGA_I2C, shows that the address pin 0 is logic 0; similarly, the connection for IC44, which is the other switch connected to MMC_I2C, shows that the address pin 0 is logic 1. Therefore, the paragraph should be corrected as:

> **FPGA I2C** and MMC I2C2, addr 1110 A2 A1 **0**
**MMC I2C1**, addr 1110 A2 A1 **1**
where A2=P2[28], A1=P2[29]
default addr is 0x70 and 0x71

HarryMakes, 2021-09-06:
An update: this typo seems to also affect Sayma_AMC. Issue: https://github.com/sinara-hw/Sayma_AMC/issues/156

filipswit, 2021-11-29:
fixed typo

closed 2021-11-29 by filipswit