sinara-hw/Metlino #73: route Fabric E

gkasprow, 2021-04-19:
At the WUT we are building a large DAQ system for cosmic muon monitoring. We use hundreds of scintillators with [SIPM ](https://hub.hamamatsu.com/jp/en/technical-note/how-sipm-works/index.html)detectors attached on both sides and [FMC-TDC](https://github.com/elhep/FMC_ADC100M_10B_TDC_16cha) boards on top of [AFCK](https://ohwr.org/projects/afck). We will use Metlino for the synchronization and control of 12 AFCK boards as well as very low latency trigger processing. Since DRTIO will use all MGTs, I plan to route LVDS to Fabric E. The new AFCK (Ultra Scale FPGA) which is under development will have an LVDS assembly option for the relevant port. In this way, we would be able to quickly transfer trigger information ( DRTIO time, TDC delay, pulse amplitude) from AFCK to Metlino using IOSERDES. It has the advantage that no CDR is required and the packet can be as short as a few bytes.
The idea is to modify Metlino slightly, without increasing the BOM at all. Samtec has [FSI-family](https://www.samtec.com/products/fsi) 10mm stacking height connectors that do not require mating parts. We simply place pads under FMC connected to Fabric E. When LVDS links are needed, we simply plug a simple FMC module with TVS protection, that links FMC with Fabric E. To further process the triggers, we will use parallel VHDCI cables to cascade more Metlinos and finally to connect to the NICA control system.

jordens, 2021-04-20:
That looks like a really cool connector for mezzanines. Few through holes, reliable registration, good density, and useful stacking height.

gkasprow, 2021-04-20:
they have many options like threaded inserts which you can use to attach screws on both sides. But that takes a lot of space. The main advantage is that it's hard to break them :)

gkasprow, 2021-04-20:
Next thing is to route 16 unused LVCMOS lines  + clock fanout output + 2 LVDS to the similar connector, but sitting below the Mordovia TONGUE 2 clock board. We will modify the Mordovia board to give access to the Metlino to the MLVDS bus. We will be able to distribute clock and 2 MLVDS signals using 1:12 LVDS fanouts. This will give us a backup solution in case of synchronization over DRTIO was not sufficient.

gkasprow, 2021-04-20:
the distance between Metlino and Mordovia is 6.03mm so there is such an option for the connector.
We need:
16 LVTTL signals for MLVDS transceivers
3 diff signals for clocks
2 signals for clock mux control
4 signals for 3V3
at least 8 signals for GND
This gives us 36 signals.
We can use the 40pin FSI-120-06-L-D-AD connector already available in many stores.

gkasprow, 2021-04-20:
@filipswit this would be your next job :)

filipswit, 2021-04-21:
sounds good ;)